OPCFoundation/UA-Nodeset #2: Decimal128 DataType

kevinherron, 2016-12-09:
What's the deal `Decimal128` in Opc.Ua.NodeSet2.Part3.xml?

Seems it's a subtype of Number, it's not a simple/alias type, but it's not specified as a built-in type? Is this something leftover that should be removed?

SparhawkSoftware, 2016-12-09:
There is an open mantis issue to add a Decimal as a built subtype of number. It should be be removed from  the published NodeSet since the WG has not finalized the updates to the spec.

kevinherron, 2016-12-09:
Is this happening in 1.04 or is it not that far along?

SparhawkSoftware, 2016-12-09:
It won't be in the RC because it lower priority than PubSub, JSON, OAuth2 and WebSocket support.
ISA95 needs it so will have to happen soon.

kevinherron, 2016-12-09:
If `Decimal128` is supposed to be a [IEEE-754 quadruple-precision](https://en.wikipedia.org/wiki/Quadruple-precision_floating-point_format) then it's going to be a pain to support in Java - there's no built-in equivalent type or parsing logic for this :(

The closest we have is the `BigDecimal` type which is an arbitrary precision signed decimal type. Seems mostly suited to being created from a string representation.

SparhawkSoftware, 2017-02-09:
Decimal type has now been official added to Part 6 1.04 RC.
It matches the Java BigDecimal type.

closed 2017-02-09 by SparhawkSoftware