OPCFoundation/UA-Nodeset #3: Remove DiagnosticsInfo

Pro, 2017-09-18:
In commit https://github.com/OPCFoundation/UA-Nodeset/blob/2e2c39ab2041ba0eecfa811aca2844d65fa3db01 the `DiagnosticsInfo(i=25)` DataType was removed from the NodeSet2.xml

In the current version, this DataType is still referenced by `StatusResult`

```
  <UADataType NodeId="i=299" BrowseName="StatusResult">
    <DisplayName>StatusResult</DisplayName>
    <References>
      <Reference ReferenceType="HasSubtype" IsForward="false">i=22</Reference>
    </References>
    <Definition Name="StatusResult">
      <Field Name="StatusCode" DataType="i=19" />
      <Field Name="DiagnosticInfo" DataType="i=25" />
    </Definition>
  </UADataType>
```

Is this an error in the nodeset?

Pro, 2017-09-18:
Ok, it wasn't this specific commit, but still `StatusResult` references the missing node `i=25`

Pro, 2017-09-18:
Additionally, the following lines reference the missing datatype `i=23` (DataValue)
```

  <UAVariable NodeId="i=3031" BrowseName="PerformInsertReplace" ParentNodeId="i=3006" DataType="i=11293">
    <DisplayName>PerformInsertReplace</DisplayName>
    <References>
      <Reference ReferenceType="HasTypeDefinition">i=68</Reference>
      <Reference ReferenceType="HasModellingRule">i=78</Reference>
      <Reference ReferenceType="HasProperty" IsForward="false">i=3006</Reference>
    </References>
  </UAVariable>
  <UAVariable NodeId="i=3032" BrowseName="NewValues" ParentNodeId="i=3006" DataType="i=23" ValueRank="1">
    <DisplayName>NewValues</DisplayName>
    <References>
      <Reference ReferenceType="HasTypeDefinition">i=68</Reference>
      <Reference ReferenceType="HasModellingRule">i=78</Reference>
      <Reference ReferenceType="HasProperty" IsForward="false">i=3006</Reference>
    </References>
  </UAVariable>
```

Pro, 2017-09-18:
I'd guess that the following definitions should be added again (they have been in the .xml before):

```
    <UADataType NodeId="i=23" BrowseName="DataValue">
        <DisplayName>DataValue</DisplayName>
        <Description>Describes a value that is a structure containing a value, a status code and timestamps.</Description>
        <References>
            <Reference ReferenceType="HasSubtype" IsForward="false">i=24</Reference>
        </References>
    </UADataType>
    <UADataType NodeId="i=25" BrowseName="DiagnosticInfo">
        <DisplayName>DiagnosticInfo</DisplayName>
        <Description>Describes a value that is a structure containing diagnostics associated with a StatusCode.</Description>
        <References>
            <Reference ReferenceType="HasSubtype" IsForward="false">i=24</Reference>
        </References>
    </UADataType>
```

bjakke, 2017-10-09:
I have raised a mantis issue of this, https://opcfoundation-onlineapplications.org/mantis/view.php?id=3980 as that should be the proper channel (however note that it requires credentials).

Pro, 2017-10-09:
Closing this in favour of the mantis issue

closed 2017-10-09 by Pro