OPCFoundation/UA-Nodeset #8: Wrong supertype for PermissionType in NodeSet2

bjakke, 2018-04-24:
The NodeSet2 xml lists PermissionType as:
```
<UADataType NodeId="i=94" BrowseName="PermissionType">
    <DisplayName>PermissionType</DisplayName>
    <References>
      <Reference ReferenceType="HasProperty">i=15030</Reference>
      <Reference ReferenceType="HasSubtype" IsForward="false">i=5</Reference>
    </References>
    <Definition Name="PermissionType" IsOptionSet="true">
      <Field Name="None" Value="0" />
      <Field Name="Browse" Value="1" />
      <Field Name="ReadRolePermissions" Value="2" />
      <Field Name="WriteAttribute" Value="4" />
      <Field Name="WriteRolePermissions" Value="8" />
      <Field Name="WriteHistorizing" Value="16" />
      <Field Name="Read" Value="32" />
      <Field Name="Write" Value="64" />
      <Field Name="ReadHistory" Value="128" />
      <Field Name="InsertHistory" Value="256" />
      <Field Name="ModifyHistory" Value="512" />
      <Field Name="DeleteHistory" Value="1024" />
      <Field Name="ReceiveEvents" Value="2048" />
      <Field Name="Call" Value="4096" />
      <Field Name="AddReference" Value="8192" />
      <Field Name="RemoveReference" Value="16384" />
      <Field Name="DeleteNode" Value="32768" />
      <Field Name="AddNode" Value="65536" />
      <Field Name="All" Value="131071" />
    </Definition>
  </UADataType>
```

Specification version 1.04 Part 3 section 8.55 lists PermissionType as subtype of UInt32, but the NodeSet lists i=5, which is UInt16. The NodeSet2 should be fixed. There is at least one other similar issue, will post a separate GH issue per problem.

bjakke, 2018-04-24:
Raised mantis issue https://opcfoundation-onlineapplications.org/mantis/view.php?id=4259

opcfoundation-org, 2018-11-23:
Fixed.

closed 2018-11-23 by opcfoundation-org