OPCFoundation/UA-Nodeset #9: Wrong supertype for EventNotifierType in NodeSet2

bjakke, 2018-04-24:
The NodeSet 2 XML lists EventNotifierType as:

```
<UADataType NodeId="i=15033" BrowseName="EventNotifierType">
    <DisplayName>EventNotifierType</DisplayName>
    <References>
      <Reference ReferenceType="HasProperty">i=15034</Reference>
      <Reference ReferenceType="HasSubtype" IsForward="false">i=7</Reference>
    </References>
    <Definition Name="EventNotifierType" IsOptionSet="true">
      <Field Name="None" Value="0" />
      <Field Name="SubscribeToEvents" Value="1" />
      <Field Name="HistoryRead" Value="4" />
      <Field Name="HistoryWrite" Value="8" />
    </Definition>
  </UADataType>
```

Specification version 1.04, Part 3 section 8.59 lists the EventNotifierType as subtype of Byte DataType, however the NodeSet2 uses i=7 which is UInt32. The NodeSet2 should be fixed.

It should be noted that this is quite important and critical in my opinion, as the EventNotifierType is listed as the DataType of the EventNotifier Attribute for Objects in version 1.04 Part 3, section 5.5.1 Table 11. In 1.03 that was Byte, so if clients/servers are made using the current NodeSet2 for 1.04 they will not use the correct DataType (they would use incorrectly the UInt32 instead of Byte, as the encoded binary form) and are incompatible with all existing applications.

bjakke, 2018-04-24:
Raised mantis issue https://opcfoundation-onlineapplications.org/mantis/view.php?id=4259

opcfoundation-org, 2018-11-23:
Fixed.

closed 2018-11-23 by opcfoundation-org