OPCFoundation/UA-Nodeset #22: UADataType.Definition.Name wrong value

mpostol, 2019-04-03:
# UADataType.Definition.Name wrong value

## Problem

The `UADataType.Definition` of the `DataTypeDefinition` type has the wrong value. For example, in the ADI model, there is `Definition`

 `<Definition Name="ExecutionCycleEnumeration">`

 According to the `Table F.12 – DataTypeDefinition` it should be `QualifiedName` - a String with the form: `<namespace index>:<name>` but it isn't as it is stated in the `Table F.2 – UANode`.

 >NOTE:
 >
 >- it affects all models
 >- it is breaking bug for the project [OPC UA NodeSet Validation](https://github.com/mpostol/OPC-UA-OOI/tree/master/SemanticData/UANodeSetValidation)

## Table F.12 – DataTypeDefinition

|Element| Type| Description
|:--:|:-:|:-|
| Name| QualifiedName|A unique name for the data type. This field is only specified for nested DataTypeDefinitions. The BrowseName of the DataType Node is used otherwise. |

## Table F.2 – UANode

|Element| Type| Description
|:--:|:-:|:-|
| BrowseName| QualifiedName|A QualifiedName serialized as a String with the form: `<namespace index>:<name>` Where the NamespaceIndex refers to the NamespaceUris table. |

## ADI snippet

``` XML
  <UADataType NodeId="ns=1;i=9378" BrowseName="1:ExecutionCycleEnumeration">
    <DisplayName>ExecutionCycleEnumeration</DisplayName>
    <References>
      <Reference ReferenceType="HasProperty">ns=1;i=13026</Reference>
      <Reference ReferenceType="HasSubtype" IsForward="false">i=29</Reference>
    </References>
    <Definition Name="ExecutionCycleEnumeration">
      <Field Name="IDLE" Value="0">
        <Description>Idle, no cleaning or acquisition cycle in progress</Description>
      </Field>
      <Field Name="DIAGNOSTIC" Value="1">
        <Description>Scquisition cycle collecting data for diagnostic purpose</Description>
      </Field>
      <Field Name="CLEANING" Value="2">
        <Description>Cleaning cycle</Description>
      </Field>
      <Field Name="CALIBRATION" Value="4">
        <Description>Calibration acquisition cycle</Description>
      </Field>
      <Field Name="VALIDATION" Value="8">
        <Description>Validation acquisition cycle</Description>
      </Field>
      <Field Name="SAMPLING" Value="16">
        <Description>Sample acquisition cycle</Description>
      </Field>
      <Field Name="DIAGNOSTIC_WITH_GRAB_SAMPLE" Value="32769">
        <Description>Scquisition cycle collecting data for diagnostic purpose and sample is extracted from the process to be sent in control lab</Description>
      </Field>
      <Field Name="CLEANING_WITH_GRAB_SAMPLE" Value="32770">
        <Description>Cleaning cycle with or without acquisition and sample is extracted from the process to be sent in control lab</Description>
      </Field>
      <Field Name="CALIBRATION_WITH_GRAB_SAMPLE" Value="32772">
        <Description>Calibration acquisition cycle and sample is extracted from the process to be sent in control lab</Description>
      </Field>
      <Field Name="VALIDATION_WITH_GRAB_SAMPLE" Value="32776">
        <Description>Validation acquisition cycle and sample is extracted from the process to be sent in control lab</Description>
      </Field>
      <Field Name="SAMPLING_WITH_GRAB_SAMPLE" Value="32784">
        <Description>Sample acquisition cycle and sample is extracted from the process to be sent in control lab</Description>
      </Field>
    </Definition>
  </UADataType>
```

## BrowseName according to the Part 3 section 8.3 QualifiedName 

This Built-in DataType contains a qualified name. It is, for example, used as BrowseName. Its elements are defined in Table 25. The name part of the QualifiedName is restricted to 512 characters.

Table 25 – QualifiedName Definition

|Element| Type| Description
|:--:|:-:|:-|
|QualifiedName| structure | 
|namespaceIndex| UInt16| Index that identifies the namespace that defines the name. This index is the index of that namespace in the local Server’s NamespaceArray. The Client may read the NamespaceArray Variable to access the string value of the namespace.
|name|String|The text portion of the QualifiedName.

mpostol, 2019-04-03:
I have also reported it as a Mantis issue:

[UADataType.Definition.Name wrong value](https://apps.opcfoundation.org/mantis/view_all_bug_page.php)

opcfoundation-org, 2019-04-03:
The text says "This field is only specified for nested DataTypeDefinitions" so the correct fix it is to delete the name from the NodeSet. 

Modellers should ignore the Name field and use the BrowseName.

mpostol, 2019-04-03:
The same table says: `The BrowseName of the DataType Node is used otherwise.`

If the correct fix it is to delete the name from the NodeSet who is in charge to delete it from officially released NodeSet files? The element is required in the schema, so it must not be just removed from the XML file?! Unfortunately, the specification says something different. 

```
  <xs:complexType name="DataTypeDefinition">
    <xs:sequence>
      <xs:element name="Field" type="DataTypeField" minOccurs="0" maxOccurs="unbounded"></xs:element>
    </xs:sequence>
    <xs:attribute name="Name" type="QualifiedName" use="required"></xs:attribute>
    <xs:attribute name="SymbolicName" type="SymbolicName" use="optional" default=""></xs:attribute>
    <xs:attribute name="IsUnion" type="xs:boolean" use="optional" default="false"></xs:attribute>
    <xs:attribute name="IsOptionSet" type="xs:boolean" use="optional" default="false"></xs:attribute>
    
    <!-- BaseType is obsolete and no longer used. Left in for backwards compatibility. -->
    <xs:attribute name="BaseType" type="QualifiedName" use="optional" default=""></xs:attribute>
  </xs:complexType>
```

My point is that we have two problems:
1. The specification and xsd is inconsitent **required** versus **This field is only specified...**
2. The content of all XML files has the wrong value - this attribute must be properly formatted. 

I am not looking for advice on how to overcome it in the tool, but for a consistent solution instead. Good practice says that the issues may be closed after getting confirmation from the reporter.

opcfoundation-org, 2019-04-04:
Update Mantis issue with discussion:
https://apps.opcfoundation.org/mantis/view.php?id=4712

mpostol, 2019-04-04:
This discussion targets the content of this repository, i.e. xsd, xml stuff, but Mantis targets the spec. In this case, both are closely coupled so some harmonization is really required. 
Because the Mantis issue recommends fixes requiring xsd and xml modifications I propose to close both issues (GitHub and Mantis) after the new version (errata, xml, xsd, etc.) has been released. It is important for developers that any tool (XML processing application, i.e. modelers, converters, servers, clients, etc. ) must apply a workaround and neglect this attribute.

opcfoundation-org, 2019-04-04:
This is wrong place for NodeSet issues.
The NodeSet is the specification and issues are tracked in Mantis by the WG.

opcfoundation-org, 2019-04-04:
Change of plan:
Will change the DataType in Part 6 from QualifiedName to String.
This will make the existing NodeSets correct.

mpostol, 2019-04-04:
I agree it is a real problem where is the right place to report issues and discuss the fix options.  Instead of classification wrong vs good, I propose useful vs official. Your proposal is not overlapping so it is a binary condition, but my proposal overlaps so it could lead to duplication. 
There is no ideal solution, hence UA WG must decide and clarify procedures. I have no vote but my point is that the issue should stay open as long as the fix will be committed to the master branch - it could be long before the official version will be available. But it is only the developer point of view.
Anyway, according to your proposal let's continue the discussion on Mantis.  So anyone interested is welcome to create the account on Mantis - fortunately, there is no need to be OPC Foundation member but be patient while looking for something relevant for you.

mpostol, 2019-04-10:
# NOTE

The mentioned above issue

- [0004712: UADataType.Definition.Name wrong value](https://apps.opcfoundation.org/mantis/view.php?id=4712)

has bee closed with the note

```
Discussed in WG call. Decisions:
1) Keep schema the same; fix model compiler;
2) Change spec to make it consistent with schema;

Outstanding issue: how manage updates to released companion specifications.
```
and cloned to a new one

- [0004724: UADataType.Definition.Name wrong value; Companion spec NodeSets need Update.](https://apps.opcfoundation.org/mantis/view.php?id=004724)

unfortunately, the discussion chain has been broken.

**My concern is that this way the syntax problem will be replaced by a semantics problem. In practice, it means that the software logic must address it instead of document validator. I understand that the price is backward compatibility.**

> I have decided to release my software without this fix so the problem is not `urgent` for me anymore.

mpostol, 2019-04-10:
I will appreciate adding the text 

`UADataType.Definition.Name wrong value #22` 

to any commit targeting this issue. It will greatly improve the problem traceability.

opcfoundation-org, 2019-04-17:
Fixed files have been posted.

mpostol, 2019-04-23:
@opcfoundation-org thanks for info. I am trying to find any relevant changes but:

1. In the ADI model `Name="ExecutionCycleEnumeration"`

```xml
<UADataType NodeId="ns=1;i=9378" BrowseName="1:ExecutionCycleEnumeration">
    <DisplayName>ExecutionCycleEnumeration</DisplayName>
    <References>
      <Reference ReferenceType="HasProperty">ns=1;i=13026</Reference>
      <Reference ReferenceType="HasSubtype" IsForward="false">i=29</Reference>
    </References>
    <Definition Name="ExecutionCycleEnumeration">
      <Field Name="IDLE" Value="0">
        <Description>Idle, no cleaning or acquisition cycle in progress</Description>
      </Field>
      ...
```
2. In the UANodeSet schema `name="Name" type="QualifiedName" use="required"`

```xml
 <xs:complexType name="UADataType">
    <xs:complexContent>
      <xs:extension base="UAType">
        <xs:sequence>
          <xs:element name="Definition" type="DataTypeDefinition" minOccurs="0"></xs:element>
        </xs:sequence>
        <xs:attribute name="Purpose" type="DataTypePurpose" default="Normal"></xs:attribute>
      </xs:extension>
    </xs:complexContent>
  </xs:complexType>

  <xs:complexType name="DataTypeDefinition">
    <xs:sequence>
      <xs:element name="Field" type="DataTypeField" minOccurs="0" maxOccurs="unbounded"></xs:element>
    </xs:sequence>
    <xs:attribute name="Name" type="QualifiedName" use="required"></xs:attribute>
    <xs:attribute name="SymbolicName" type="SymbolicName" use="optional" default=""></xs:attribute>
    <xs:attribute name="IsUnion" type="xs:boolean" use="optional" default="false"></xs:attribute>
    <xs:attribute name="IsOptionSet" type="xs:boolean" use="optional" default="false"></xs:attribute>
    
    <!-- BaseType is obsolete and no longer used. Left in for backwards compatibility. -->
    <xs:attribute name="BaseType" type="QualifiedName" use="optional" default=""></xs:attribute>
  </xs:complexType>
```
3. I cannot find also any relevant changes in the logs chain.

> **Note**:

```xml
  <xs:simpleType name="QualifiedName">
    <xs:restriction base="xs:string"></xs:restriction>
  </xs:simpleType>
```
`Name="ExecutionCycleEnumeration"` is valid against the schma but is not valid against the specyfication.  According to the specyfication definition of the `QualifiedName` schould have additional restictions. 

> I will appreciate any more accurate information about what has been changed. 

My concern is that the schema must be compliant with the specification but it is not. 

Hope it helps.

opcfoundation-org, 2019-04-23:
Sorry for not being explicit:
ADI was updated in the v1.03 branch:
https://github.com/OPCFoundation/UA-Nodeset/blob/v1.03/ADI/Opc.Ua.Adi.NodeSet2.xml

mpostol, 2019-04-23:
❗️  
- `Showing  with 6,291 additions and 4,614 deletions.`
- 44 changed files (not sure if all existing models have been changed)
- cannot be merged to master
- 1.03 has been already released? Next is 1.03.04 bus following the [Semantic Versioning 2.0.0](https://semver.org/) it should be 2.0
- How this stuff is related to <https://opcfoundation.org/UA/schemas/1.04/> that has been just posted?

> My point is that as a result of this modification, a new restriction should be added as I proposed above. 

Anyway, to coordinate our work, my plan is as follows:

1, try to figure out a migration path - how to apply all these changes (probably I need spec errata)
2. try to validate the new files against the semantics defined in Part 3, 5, and 6. using the [

Can you advise me what information in the xml file is best suited to be used for switching the syntax parser?

Many thanks for the help.

opcfoundation-org, 2019-04-23:
The older models were problematic because rebuilding them with 1.04 introduces other changes.
The 1.03 branch was updated with the old models built against the 1.03 nodeset for this reason.
You should not try to merge 1.03 into master. The ADI in that master should built against the latest NodeSet which is not the official release for ADI.

The WG decided that no changes to the schema or the nodeset would be made. Only that old models will be fixed.

mpostol, 2019-04-23:
To be honest I don't understand. Hopefully, WG knows much more and are able to predict precisely consequences. The only thing I understand is the necessity to change my plans to do nothing.
Many thanks for the cooperation. It seems that's all we can do now except closing this issue.

closed 2019-05-02 by randy-armstrong