OPCFoundation/UA-Nodeset #25: Wrong Reference in DI Spec XML?

biancode, 2019-04-28:
Hi,

<Reference ReferenceType="HasSubtype" IsForward="false">i=17602</Reference>

The i=17602 Reference is just a HasSubType, but is unknown on importing XML as a type self from the XML. With that modeler will not import that file.

kind regards,
Klaus

biancode, 2019-04-28:
unknown 17603

```
      <Reference ReferenceType="i=17603">ns=1;i=15036</Reference>
      <Reference ReferenceType="i=17603">ns=1;i=15048</Reference>
```

biancode, 2019-04-28:
from the new ObjectType:
`<UAObjectType NodeId="ns=1;i=15062" BrowseName="1:ComponentType" IsAbstract="true">`

biancode, 2019-04-28:
BaseInterfaceType,17602,ObjectType
HasInterface,17603,ReferenceType

randy-armstrong, 2019-05-02:
The readme for the repo was updated to communicate the process that is currently used.
If you are looking for DI 1.1 you must pull it from the v1.03 branch.
The master and v1.04 branches pull in new types which require tools that handle later versions of the OPC UA specification.

closed 2019-06-13 by randy-armstrong