OPCFoundation/UA-Nodeset #26: UANodeSet on website and in the repo branch 1.04 are different

mpostol, 2019-05-06:
I am not sure which one I should use as the current 1.04 release.

The file UANodeSet in the repo 1.04 starts 

```XML
<UANodeSet xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" LastModified="2019-05-01T00:00:00Z" xmlns="http://opcfoundation.org/UA/2011/03/UANodeSet.xsd">
  <Models>
    <Model ModelUri="http://opcfoundation.org/UA/" Version="1.04" PublicationDate="2019-05-01T00:00:00Z" />
````
on the website pointed out by the readme.md <https://opcfoundation.org/UA/schemas/1.04/>
it looks like this

````XML
<UANodeSet xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" LastModified="2019-04-15T00:00:00Z" xmlns="http://opcfoundation.org/UA/2011/03/UANodeSet.xsd">
  <Models>
    <Model ModelUri="http://opcfoundation.org/UA/" Version="1.04" PublicationDate="2019-04-15T00:00:00Z" />
````

> **Unfortunately, I cannot compare both because of the huge number of differences. **
> Which one has been released? Both are marked `Version="1.04`. 

Usually, the GitHub repositories contain source files. This repository contains compiler results. It could be the cause of the tracking problem.  New compilation could produce different results. Additionally XML files make the diff process more challenging or even impossible to manage because two different XML files don't mean different information they represent. Anyway, special care must be taken to keep all the staff in sync and precise text description of releases to avoid any confusion. 

Why *.xml files are in the schema folder?

mpostol, 2019-05-06:
Consider releasing the relevant UANodeSet files as, for example,  the NuGet packages or compiled into the resource files(or similar depending on the technology). This way versioning could be managed much easier.

randy-armstrong, 2019-06-13:
The readme explains the process:
https://github.com/OPCFoundation/UA-Nodeset

There is one updated agreed by the WG at the F2F:
From now on there will be an errata published each time the version on the website changes.
The version in GitHub will continue to be a version used during the review process.

mpostol, 2019-06-14:
OK, I will try to follow these rules. Let's see how it works in practice.

closed 2019-06-14 by mpostol