OPCFoundation/UA-Nodeset #27: Problem when importing DI nodeset version 1.02 from branch 1.04, parent node id is sometimes omitted

Oidlichtnwoada, 2019-06-04:
For example the node <UAObjectType NodeId="ns=1;i=15035" BrowseName="1:IVendorNameplateType" ParentNodeId="" IsAbstract="true">
    <DisplayName>IVendorNameplateType</DisplayName>
    <References>
      <Reference ReferenceType="HasProperty">ns=1;i=15036</Reference>
      <Reference ReferenceType="HasProperty">ns=1;i=15037</Reference>
      <Reference ReferenceType="HasProperty">ns=1;i=15038</Reference>
      <Reference ReferenceType="HasProperty">ns=1;i=15039</Reference>
      <Reference ReferenceType="HasProperty">ns=1;i=15040</Reference>
      <Reference ReferenceType="HasProperty">ns=1;i=15041</Reference>
      <Reference ReferenceType="HasProperty">ns=1;i=15042</Reference>
      <Reference ReferenceType="HasProperty">ns=1;i=15043</Reference>
      <Reference ReferenceType="HasProperty">ns=1;i=15044</Reference>
      <Reference ReferenceType="HasProperty">ns=1;i=15045</Reference>
      <Reference ReferenceType="HasProperty">ns=1;i=15046</Reference>
      <Reference ReferenceType="HasProperty">ns=1;i=15047</Reference>
      <Reference ReferenceType="HasSubtype" IsForward="false">i=17602</Reference>
    </References>
  </UAObjectType> has no parent node id. Therefore the importer cannot dynamically recreate the nodeset. What is the default parent node id used in this nodeset?

ccvca, 2019-06-05:
I would suggest, that the parent node is the source of the inverse refrence, which is defined here:
https://github.com/OPCFoundation/UA-Nodeset/blob/0777abd1bc407b4dbd79abc515864f8c3ce6812b/DI/Opc.Ua.Di.NodeSet2.xml#L551

Oidlichtnwoada, 2019-06-12:
Thanks, that solved the problem

closed 2019-06-12 by Oidlichtnwoada