OPCFoundation/UA-Nodeset #29: Why are ArrayDimensions = 0 in the NodeSet files?

mlang-de, 2019-07-25:
In most cases the ArrayDimension is know for instance for InputArguments. Is there a reason why the ArrayDimensions is 0 all time?
Checked NodeSet file are BaseNodeSet and DI.

randy-armstrong, 2019-07-25:
ArrayDimensions="0" means the 1 dimensional array can have any length.
It is only valid for ValueRank=1.
If it is specified for ValueRank != 1 then it is an error that needs fixing.

mlang-de, 2019-07-25:
Here is my exmaple:
[image]
Why is here ArrayDimension="0" used and not "2"?
Imho both are permitted.

randy-armstrong, 2019-07-31:
ValueRank="1" in your example. The NodeSet appears to be correct.

mlang-de, 2019-07-31:
The length of the ExtensionObject array is known, so why is this not indicated by the ArrayDimension with "2"? Regardless that "0" is also valid.

opcfoundation-org, 2019-08-09:
The ArrayDimensions specifies a constraint on the Value of the Node. It does NOT indicate the length of the current value.

mlang-de, 2019-08-09:
This is true, but in the example above it could be constrained, because the length is fixed by the specification.

opcfoundation-org, 2019-08-10:
That is true.

opcfoundation-org, 2020-11-26:
This is fixed in the version in OPCFMembers repo.
Will be published in the public repo in the next release.

closed 2021-01-27 by opcfoundation-org