OPCFoundation/UA-Nodeset #31: Problems when using DI model version 1.1 (specified in companion spec) with his UA model required version

Nik654, 2019-07-30:
## Description
In OPC Unified Architecture for Devices Companion Specification Release 1.01 specifies that model Schema can be found on link /https://opcfoundation.org/UA/schemas/DI/1.1/. 
Here we have DI model with publication date "2018-06-07T00:00:00Z" and it requires UA model version 1.03 with publication date "2013-12-02T00:00:00Z". When i try combine UA model with publication date "2013-12-02T00:00:00Z" (https://opcfoundation.org/UA/schemas/1.03/Opc.Ua.NodeSet2.xml) and DI model specified in companion specification (https://opcfoundation.org/UA/schemas/DI/1.1/) i get errors.
This is because DI model is requesting some datatypes which are not available in UA model. E.g. line from https://opcfoundation.org/UA/schemas/DI/1.1/Opc.Ua.Di.NodeSet2.xml:
UAVariable NodeId="ns=1;i=15031" BrowseName="DefaultRolePermissions" ParentNodeId="ns=1;i=15001" DataType="i=96" ValueRank="1" ArrayDimensions="0"

DataType="i=96" is looked for in UA model, but doesn't exist there and because of this i am having problems with these models.

Should maybe required version in DI Opc.Ua.Di.NodeSet2.xml file be UA version 1.04 or maybe i have misunderstood something?

randy-armstrong, 2019-07-31:
Yes, the reference should be updated to point to 1.04.
Older 1.03 compatible versions of the NodeSet can be found here:
https://github.com/OPCFoundation/UA-Nodeset/blob/v1.03/DI/Opc.Ua.Di.NodeSet2.xml

Nik654, 2019-07-31:
One more question, when it says in .xml file, <RequiredModel ModelUri=... , does this mean:
1. **Minimal** required model and than you can use same version but just newer model (by PublicationDate)
2. **Only** set required version must be used (no newer versions allowed)

You can than close this issue when fix is implemented if this is OK for you?

opcfoundation-org, 2019-08-09:
The requirement is a minimum since the general policy is newer versions of the NodeSet are always backwards compatible unless there was a serious error that meant a feature was broken and unusable.

Nik654, 2019-08-09:
Thank you for the info.

closed 2019-10-12 by randy-armstrong