OPCFoundation/UA-Nodeset #43: Problems with Robotics CS Nodeset

andre-fm, 2019-09-25:
Hello,

I am experiencing some problems when trying to import the "Opc.Ua.Robotics.NodeSet2.xml" and "Opc.Ua.Di.NodeSet2.xml" to UA Modeler software from Unified Automation.
With "modeling only" purpose, when importing the two .xml files I get this kind of errors "TypeDefinition not found (<number>)".
[image]
[image]
Is there anyone with the same kind of problem?

Best Regards,
André Martins

xenonforlife, 2019-09-25:
Hi,
The Robotics CS makes use of some Type definitions from Amendment 1 and Amendment 11. However these (new) type definitions from the Ammendments have not yet been incorporated in the NodeSet files of the core standard. Therefore it results in such errors. 
These errors will all be taken care of with a new version of UaModeler which is soon to be released (as far as I know), since it would come with the updated core NodeSet files.
I hope this answers your query :)

Thanks,
Sup

andre-fm, 2019-09-25:
Thanks for your comment . Then I will have to wait for the new version of UaModeler. :)

Regards,
André Martins

closed 2020-01-30 by opcfoundation-org

Ayyubshaikh, 2021-07-04:
I am Also not able to add robotics nodeset file to UAModeler Models. Is it possible now to do it???

GoetzGoerisch, 2021-07-04:
@Ayyubshaikh what errors do you see? I'm using this nodeset without issues in the UaModeler. But a new version is in the works to fix some other errors.

Ayyubshaikh, 2021-07-04:
@GoetzGoerisch . I was doing it wrong. I figured it out. 
Does it look fine.
Thankyou
[image]