OPCFoundation/UA-Nodeset #49: ISA95 model - missing Opc.ISA95.Model.xml file

rudoq32, 2019-11-09:
Hello @randy-armstrong,
I found this repository as a better place for this question as  [OPCFoundation/UA-ModelCompiler #37](https://github.com/OPCFoundation/UA-ModelCompiler/issues/37).
You mentioned that the file **Opc.ISA95.Model.xml** should be provided by work group... I am open for contributing in this action if it could accelerate the process.
Thank you 
Best regards Rudo

opcfoundation-org, 2020-08-28:
If you still want to contribute you need to contact the WG chair:
https://docs.google.com/spreadsheets/d/10SOpad6uu7JA5ZSpccVyqaqkyhYBiIXqNus28-1cJtU/edit#gid=1248333029

closed 2020-08-28 by opcfoundation-org

STUBA-rupr, 2020-09-13:
Dear @opcfoundation-org,
in the column contact for ISA95 is <AskOPC>, to whom I could address my email?
Thank you
R.