OPCFoundation/UA-Nodeset #54: New Object "Aliases" in the Opc.Ua.NodeSet2.xml

prrw, 2020-02-27:
In the latest commit of the `Opc.Ua.NodeSet2.xml`, there is an inclusion of a new Object "Aliases". Why is it included or is it a bug ?
```
<UAObject NodeId="i=23470" BrowseName="Aliases" ReleaseStatus="Draft">
    <DisplayName>Aliases</DisplayName>
    <References>
      <Reference ReferenceType="HasComponent">i=23476</Reference>
      <Reference ReferenceType="Organizes" IsForward="false">i=85</Reference>
      <Reference ReferenceType="HasTypeDefinition">i=23456</Reference>
    </References>
  </UAObject>
  <UAMethod NodeId="i=23476" BrowseName="FindAlias" ReleaseStatus="Draft" ParentNodeId="i=23470" MethodDeclarationId="i=23462">
    <DisplayName>FindAlias</DisplayName>
    <References>
      <Reference ReferenceType="HasProperty">i=23477</Reference>
      <Reference ReferenceType="HasProperty">i=23478</Reference>
      <Reference ReferenceType="HasComponent" IsForward="false">i=23470</Reference>
    </References>
  </UAMethod>
  <UAVariable NodeId="i=23477" BrowseName="InputArguments" ReleaseStatus="Draft" ParentNodeId="i=23476" DataType="i=296" ValueRank="1" ArrayDimensions="0">
    <DisplayName>InputArguments</DisplayName>
    <References>
      <Reference ReferenceType="HasTypeDefinition">i=68</Reference>
      <Reference ReferenceType="HasProperty" IsForward="false">i=23476</Reference>
    </References>
    <Value>
      <ListOfExtensionObject xmlns="http://opcfoundation.org/UA/2008/02/Types.xsd">
        <ExtensionObject>
          <TypeId>
            <Identifier>i=297</Identifier>
          </TypeId>
          <Body>
            <Argument>
              <Name>AliasNameSearchPattern</Name>
              <DataType>
                <Identifier>i=12</Identifier>
              </DataType>
              <ValueRank>-1</ValueRank>
              <ArrayDimensions />
            </Argument>
          </Body>
        </ExtensionObject>
        <ExtensionObject>
          <TypeId>
            <Identifier>i=297</Identifier>
          </TypeId>
          <Body>
            <Argument>
              <Name>ReferenceTypeFilter</Name>
              <DataType>
                <Identifier>i=17</Identifier>
              </DataType>
              <ValueRank>-1</ValueRank>
              <ArrayDimensions />
            </Argument>
          </Body>
        </ExtensionObject>
      </ListOfExtensionObject>
    </Value>
  </UAVariable>
  <UAVariable NodeId="i=23478" BrowseName="OutputArguments" ReleaseStatus="Draft" ParentNodeId="i=23476" DataType="i=296" ValueRank="1" ArrayDimensions="0">
    <DisplayName>OutputArguments</DisplayName>
    <References>
      <Reference ReferenceType="HasTypeDefinition">i=68</Reference>
      <Reference ReferenceType="HasProperty" IsForward="false">i=23476</Reference>
    </References>
    <Value>
      <ListOfExtensionObject xmlns="http://opcfoundation.org/UA/2008/02/Types.xsd">
        <ExtensionObject>
          <TypeId>
            <Identifier>i=297</Identifier>
[... 165 lines cut ...]
    <Value>
      <ListOfExtensionObject xmlns="http://opcfoundation.org/UA/2008/02/Types.xsd">
        <ExtensionObject>
          <TypeId>
            <Identifier>i=297</Identifier>
          </TypeId>
          <Body>
            <Argument>
              <Name>AliasNodeList</Name>
              <DataType>
                <Identifier>i=23468</Identifier>
              </DataType>
              <ValueRank>1</ValueRank>
              <ArrayDimensions>
                <UInt32>0</UInt32>
              </ArrayDimensions>
            </Argument>
          </Body>
        </ExtensionObject>
      </ListOfExtensionObject>
    </Value>
  </UAVariable>
```
It makes my server look this way.
[image]

PaulHunkar, 2020-03-01:
The Aliases node is a new addition that is in part 17, I think it might be early, in that it is only an RC, but the definition appears to be correct and it should move to released some time fairly soon.

prrw, 2020-03-02:
Okay Thanks. I will check that out.

closed 2020-03-02 by prrw

mpostol, 2020-03-02:
Hi @PaulHunkar,

My concern is that using the object-oriented approach we should use inheritance as the paradigm to expand the existing model.  Adding, over and over new stuff makes the model unstable and inevitably causes versioning problems.  As a consequence, this could put in danger the interoperability, which must be the main concern of the standardization process. 

Partially I know the answer.  @PaulHunkar  let me recall our discussion many years ago about the fundamental difference between:

- OPC UA Information Model definition - we can carve it in stone
- OPC UA Address Space instance - we need the power supply to keep it alive

Between both there is the ModelCompiler tool. Inheritance makes sense only in the case of the `Information Model` - "one stone (derived) can refer to another one (base)", so there is no need to reshape the base one. Talking about `NodeSet2` we are talking about a formal description of the Address Space instance.

> What do you think about returning to this discussion to avoid problems like this in the future? 

I am not sure what exactly means "I will check that out" (@prv), but I know many cases where this means I will keep it on hold for the next couple years :(.

The discussion is important for me because as a part of the lab: Programming Technologies - Mobile Devices we will try to use [**Blockchain**](https://github.com/mpostol/NBlockchain#nblockchain) as the repository of the custom OPC UA Information Model definitions.

Any comments are welcome. 

Mariusz