OPCFoundation/UA-Nodeset #60: Nodes don't have Reference to its ParentNodeId

BigeYoung, 2020-06-03:
I've checked the whole repository and found some nodes don't have any kind of `Reference` to its `ParentNodeId`.

This caused FreeOpcUa to raise `BadParentNodeIdInvalid` Exception, but I don't know if it is FreeOpcUa's issue ([#1056](https://github.com/FreeOpcUa/python-opcua/issues/1056)) or yours.

---
All the nodes that don't have Reference to its ParentNodeId can be found here.
Most of them are `UAVariable` from **AML** directory but two of them are `UAObject` from **ISA-95** directory.

### UAVariable from AML
[AML/Opc.Ua.AMLBaseTypes.NodeSet2.xml](https://github.com/OPCFoundation/UA-NodesetUA-Nodeset-master/AML/Opc.Ua.AMLBaseTypes.NodeSet2.xml)
['ns=1;i=1010', 'ns=1;i=1011']
[AML/Opc.Ua.AMLLibraries.NodeSet2.xml](https://github.com/OPCFoundation/UA-NodesetUA-Nodeset-master/AML/Opc.Ua.AMLLibraries.NodeSet2.xml)
['ns=2;i=2', 'ns=2;i=3', 'ns=2;i=5', 'ns=2;i=6', 'ns=2;i=8', 'ns=2;i=9', 'ns=2;i=11', 'ns=2;i=12', 'ns=2;i=14', 'ns=2;i=15', 'ns=2;i=17', 'ns=2;i=18', 'ns=2;i=20', 'ns=2;i=21', 'ns=2;i=23', 'ns=2;i=24', 'ns=2;i=26', 'ns=2;i=27', 'ns=2;i=29', 'ns=2;i=30', 'ns=2;i=32', 'ns=2;i=33', 'ns=2;i=35', 'ns=2;i=36', 'ns=2;i=38', 'ns=2;i=39', 'ns=2;i=41', 'ns=2;i=42', 'ns=2;i=44', 'ns=2;i=45', 'ns=2;i=47', 'ns=2;i=48', 'ns=2;i=50', 'ns=2;i=51', 'ns=2;i=53', 'ns=2;i=54', 'ns=2;i=56', 'ns=2;i=57', 'ns=2;i=59', 'ns=2;i=60', 'ns=2;i=62', 'ns=2;i=63', 'ns=2;i=65', 'ns=2;i=66', 'ns=2;i=68', 'ns=2;i=69', 'ns=2;i=70', 'ns=2;i=72', 'ns=2;i=73', 'ns=2;i=75', 'ns=2;i=76', 'ns=2;i=78', 'ns=2;i=79', 
'ns=2;i=81', 'ns=2;i=82', 'ns=2;i=84', 'ns=2;i=85', 'ns=2;i=87', 'ns=2;i=88', 'ns=2;i=90', 'ns=2;i=91', 'ns=2;i=93', 'ns=2;i=94', 'ns=2;i=96', 'ns=2;i=97', 'ns=2;i=99', 'ns=2;i=100', 'ns=2;i=102', 'ns=2;i=103', 'ns=2;i=105', 'ns=2;i=106', 'ns=2;i=108', 'ns=2;i=109', 'ns=2;i=111', 'ns=2;i=112', 'ns=2;i=114', 'ns=2;i=115', 'ns=2;i=117', 'ns=2;i=118', 'ns=2;i=120', 'ns=2;i=121', 'ns=2;i=123', 'ns=2;i=124', 'ns=2;i=126', 'ns=2;i=127', 'ns=2;i=129', 'ns=2;i=130', 'ns=2;i=132', 'ns=2;i=133', 'ns=2;i=135', 'ns=2;i=136', 'ns=2;i=138', 'ns=2;i=139', 'ns=2;i=141', 'ns=2;i=142', 'ns=2;i=144', 'ns=2;i=145', 'ns=2;i=147', 'ns=2;i=148', 'ns=2;i=150', 'ns=2;i=151', 'ns=2;i=153', 'ns=2;i=154', 'ns=2;i=156', 'ns=2;i=157', 'ns=2;i=159', 'ns=2;i=160', 'ns=2;i=162', 'ns=2;i=163', 'ns=2;i=165', 'ns=2;i=166', 'ns=2;i=168', 'ns=2;i=169', 'ns=2;i=171', 'ns=2;i=172', 'ns=2;i=174', 'ns=2;i=175', 'ns=2;i=177', 'ns=2;i=178', 'ns=2;i=180', 'ns=2;i=181', 'ns=2;i=183', 'ns=2;i=184', 'ns=2;i=186', 'ns=2;i=187', 'ns=2;i=189', 'ns=2;i=190', 'ns=2;i=192', 'ns=2;i=193', 'ns=2;i=195', 'ns=2;i=196', 'ns=2;i=198', 'ns=2;i=199', 'ns=2;i=201', 'ns=2;i=202', 'ns=2;i=204', 'ns=2;i=205', 'ns=2;i=207', 'ns=2;i=208', 'ns=2;i=210', 'ns=2;i=211', 'ns=2;i=213', 'ns=2;i=214', 'ns=2;i=216', 'ns=2;i=217', 'ns=2;i=219', 'ns=2;i=220', 'ns=2;i=222', 'ns=2;i=223', 'ns=2;i=225', 'ns=2;i=226', 'ns=2;i=228', 'ns=2;i=229', 'ns=2;i=231', 'ns=2;i=232', 'ns=2;i=234', 'ns=2;i=235', 'ns=2;i=237', 'ns=2;i=238', 'ns=2;i=240', 'ns=2;i=241', 'ns=2;i=243', 'ns=2;i=244', 'ns=2;i=246', 'ns=2;i=247', 'ns=2;i=249', 'ns=2;i=250', 'ns=2;i=252', 'ns=2;i=253', 'ns=2;i=255', 'ns=2;i=256', 'ns=2;i=258', 'ns=2;i=259', 'ns=2;i=261', 'ns=2;i=262', 'ns=2;i=264', 'ns=2;i=265', 'ns=2;i=267', 'ns=2;i=268', 'ns=2;i=270', 'ns=2;i=271', 'ns=2;i=273', 'ns=2;i=274', 'ns=2;i=276', 'ns=2;i=277', 'ns=2;i=279', 'ns=2;i=280', 'ns=2;i=282', 'ns=2;i=283', 'ns=2;i=285', 'ns=2;i=286', 'ns=2;i=288', 'ns=2;i=289', 'ns=2;i=291', 'ns=2;i=292', 'ns=2;i=294', 'ns=2;i=295', 'ns=2;i=297', 'ns=2;i=298', 'ns=2;i=300', 'ns=2;i=301', 'ns=2;i=303', 'ns=2;i=304', 'ns=2;i=306', 'ns=2;i=307', 'ns=2;i=309', 'ns=2;i=310', 'ns=2;i=312', 'ns=2;i=313', 'ns=2;i=315', 'ns=2;i=316', 'ns=2;i=318', 'ns=2;i=319', 'ns=2;i=321', 'ns=2;i=322', 'ns=2;i=324', 'ns=2;i=325', 'ns=2;i=327', 'ns=2;i=328', 'ns=2;i=330', 'ns=2;i=331', 'ns=2;i=333', 'ns=2;i=334', 'ns=2;i=336', 'ns=2;i=337', 'ns=2;i=339', 'ns=2;i=340', 'ns=2;i=341']
[AML/Topology.xml](https://github.com/OPCFoundation/UA-NodesetUA-Nodeset-master/AML/Topology.xml)
['ns=2;i=3', 'ns=2;i=5', 'ns=2;i=7', 'ns=2;i=9', 'ns=2;i=11', 'ns=2;i=14', 'ns=2;i=16', 'ns=2;i=21', 'ns=2;i=23', 'ns=2;i=24']

### UAObject from ISA-95
[ISA-95/Opc.ISA95.NodeSet2.xml](https://github.com/OPCFoundation/UA-NodesetUA-Nodeset-master/ISA-95/Opc.ISA95.NodeSet2.xml)
['ns=1;i=5224', 'ns=1;i=5295']

BigeYoung, 2020-06-03:
The trouble is, there are many kinds of parent nodes in this situation, and some parent nodes also don't have reference to its child node.

opcfoundation-org, 2020-06-08:
These do not look like errors to me.
There is no requirement that both directions of a reference be defined.
See https://reference.opcfoundation.org/v104/Core/docs/Part6/F.3/

> A UANodeSet is expected to contain many UANodes which reference each other. Tools that create UANodeSets should not add Reference elements for both directions in order to minimize the size of the XML file. Tools that read the UANodeSets shall automatically add reverse references unless reverse references are not appropriate given the ReferenceType semantics. HasTypeDefinition and HasModellingRule are two examples where it is not appropriate to add reverse references.

BigeYoung, 2020-06-09:
> These do not look like errors to me.
> There is no requirement that both directions of a reference be defined.
> See https://reference.opcfoundation.org/v104/Core/docs/Part6/F.3/

Thank you for answering my question.   

Now I know that most of the nodes I mentioned are no problem. But I still have a little question, the two nodes from AML/Opc.Ua.AMLBaseTypes.NodeSet2.xml dont' have references to their parent node, and their parent node don't have references to them, neither. Maybe this is an error?

AML/Opc.Ua.AMLBaseTypes.NodeSet2.xml
['ns=1;i=1010', 'ns=1;i=1011']

opcfoundation-org, 2020-06-16:
AMLBaseVariableType has references to those nodes.

closed 2020-06-30 by opcfoundation-org