OPCFoundation/UA-Nodeset #65: whenever I try to run any model I get the error like the following

NakWarsi, 2020-07-17:
`failure adding node NodeData(nodeid:NumericNodeId(ns=1;i=1001))
Traceback (most recent call last):
  File "generic.py", line 17, in <module>
  File "site-packages\opcua\server\server.py", line 510, in import_xml
  File "site-packages\opcua\common\xmlimporter.py", line 67, in import_xml
  File "site-packages\opcua\common\xmlimporter.py", line 84, in _add_node_data
  File "site-packages\opcua\common\xmlimporter.py", line 196, in add_object_type
  File "site-packages\opcua\ua\uatypes.py", line 218, in check
opcua.ua.uaerrors._auto.BadParentNodeIdInvalid: "The parent node id does not to refer to a valid node."(BadParentNodeIdInvalid)
[18060] Failed to execute script generic`

are these models runnable with OpcUa Servers?

swamper123, 2020-07-17:
You may post this question in the python-opcua repo (I guess you are using their code) and check out their schemas folder as well.

NakWarsi, 2020-07-17:
but the server is running the samples I have created and few of the model from this repo also that's why I thought maybe error is there in the information models only,
and @swamper123  what you meant by checking their schema folder?

swamper123, 2020-07-17:
The standard `server()` of [python-opcua](https://github.com/FreeOpcUa/python-opcua) and [opcua-asyncio](https://github.com/FreeOpcUa/opcua-asyncio) use the latest public nodeset.
Try to start that one, before you import your own nodes.
If this works well, then you add your "extra Nodes"-Set, **before** you execute` server.start()`, via `server.import_xml("my_own_made_extra_nodeset.xml"`).

If this fails, you know something is wrong with your nodeset.

Since this Nodeset is used since a couple of weeks and nobody mentioned a problem in both python repos, I guess this issue can be closed here (and in case something still went wrong, open an issue there. If realy everything seems to falling apart, we can open a new issue here again 😉).

NakWarsi, 2020-07-17:
thanks, @swamper123

closed 2020-07-17 by NakWarsi