OPCFoundation/UA-Nodeset #75: OPC UA Robotics - bsd file

aashik-udupi, 2020-10-27:
The _OPC UA Companion Spec for Robotics_ defines new **Enumeration DataTypes** 
-  AxisMotionProfileEnumeration _(used under AxisType)_
- ExecutionModeEnumeration _(used under TaskControlType)_ 
- MotionDeviceCategoryEnumeration _(used under MotionDeviceType)_ 
- OperationalModeEnumeration _(used under SafetyStateType)_ 

However, no official **bsd** file was released as part of the OPC UA Robotics Schema. Any reason why?

GoetzGoerisch, 2020-10-30:
@xenonforlife could you please help here?

audupi, 2020-10-30:
In Comparison with the OPCUA DI bsd, I have created the Robotics.bsd based on data types mentioned above. 

I have attached the same as text file here. It would be great if anyone could affirm this! 

Thanks in advance!

[Opc.Ua.Robotics.Types.bsd.txt](https://github.com/OPCFoundation/UA-Nodeset/files/5464797/Opc.Ua.Robotics.Types.bsd.txt)

opcfoundation-org, 2020-10-30:
The BSD is embedded in the NodeSet as Base64 data.
It can be extracted with a copy and paste.
We have requested that all specification writers provide the BSD when their spec is released.
It will take a bit of time for current specifications to catch up.

closed 2020-11-26 by opcfoundation-org