OPCFoundation/UA-Nodeset #80: Problem on ISA-95 nodeset file with open62541

hanyangyexun, 2021-04-12:
Hello,
I was trying to use the nodeset-compiler function to compile the ISA-95 nodeseet (from OPC foundation nodeset).

According to the open62541 manual, every step was successful but there was error when executing the final OPC UA server (-o file).

I also tried the same process for compile the IEC 61850-7-3 and the MDIS nodeset, they are successful.

This is strange and it is hard for me to locate the problem.

Did anyone have experience on nodeset-compiler?

My guess was that the ISA-95 nodeset file was outdated as it was developed a long time ago in 2013, the open62541 cannot parse it now.

Thank you.

GoetzGoerisch, 2021-04-12:
The [open62541](https://github.com/open62541/open62541) repo would be the better place to ask this question.

Please check out a sample implementation for machine tools and robotics companion spec here: [Sample-Server](https://github.com/umati/Sample-Server/blob/main/CMakeLists.txt)

Normally the _NodeSets_ are parsed with each CI build of [open62541](https://github.com/open62541/open62541).

br-iosb, 2021-11-15:
The problem could arise when parsing the NodeId csv. To my knowledge, in I4AAS there are some names which differ only case-sensitively and open62541 reads these in upper-case, which leads to ambiguities.

closed 2026-10-21 by opcfoundation-org