OPCFoundation/UA-Nodeset #85: CNC Nodeset cannot access the xml

JinhwanPark, 2021-07-27:
Hello
I am trying to use CNC Nodeset.
The CNC document address of the current OPC Foundation is as follows:
http://www.opcfoundation.org/UA/CNC/1.0/Opc.Ua.CNC.NodeSet.xml
I cannot access the xml file with that address.

GoetzGoerisch, 2021-07-28:
You can find the NodeSet also here: https://github.com/OPCFoundation/UA-Nodeset/tree/v1.04/CNC

Please be aware, there are a few pending Mantis Issues with the companion specification and the working group is currently updating the specification, which might lead to breaking changes.

I would be happy if you join the joint working group and contribute your experience in this process. Please get in touch be mail:
https://vdw.de/der-vdw/ansprechpartner/#technikforschung

opcfoundation-org, 2021-11-14:
This error looks like an error in the FreeOpcUa parser. 
I can provide suggestions without knowing the exact line that was causing the error.
The original link is fixed.

closed 2022-05-27 by opcfoundation-org