OPCFoundation/UA-Nodeset #86: 1:AASDataTypeIEC61360DataType in I4AAS nodeset, wrong EnumString

janfelixklein, 2021-11-01:
Hello everyone,

I will leave an issue here since it looks like the I4AAS is not available for bug tracking in mantis (at least I couldn't find it).

I found the following problem with the spec:

`UADataType NodeId="ns=1;i=3008" BrowseName="1:AASDataTypeIEC61360DataType"` specifies a total number of 15 DataTypes, however its EnumString `ns=1;i=6111` property is not alligned and only specifies 11 types. 

From my understanding, both should be referring to the same types

GoetzGoerisch, 2021-11-01:
@dk2129 please open a Mantis project.
@fpethig could you help on this?

janfelixklein, 2021-11-11:
I still don't see any Mantis project, I also found some additional problems with the specs

swamper123, 2021-11-11:
@janfelixklein You may open an issue on http://mantis.opcfoundation.org which is the offical bug tracker of the foundation ([see here](https://github.com/OPCFoundation/UA-Nodeset#error-reporting))

You can mention even more problems of a spec there.

opcfoundation-org, 2021-11-18:
Working on this.

janfelixklein, 2021-12-06:
I just checked on mantis again. There is still no project for the I4AAS specification...

opcfoundation-org, 2021-12-06:
Sorry, it is not just creating the project - we have to find the right person to assign as manager and make sure they are aware that they need to deal with mantis issues. Send another follow up email.

opcfoundation-org, 2021-12-10:
Sorry for the delay:
https://mantis.opcfoundation.org/set_project.php?project_id=155&amp;make_default=no

closed 2026-10-21 by opcfoundation-org