OPCFoundation/UA-Nodeset #94: Eumabois nodeset requires non-existent version of Woodworking standard

jacobq, 2023-04-06:
What am I missing here? It appears that [Woodworking/Opc.Ua.Eumabois.Nodeset2.xml](https://github.com/OPCFoundation/UA-Nodeset/blob/latest/Woodworking/Opc.Ua.Eumabois.Nodeset2.xml) references version `1.00.8` of the `Woodworking` companion specification:
https://github.com/OPCFoundation/UA-Nodeset/blob/526c5e59e7ba9e54f9dc7848ce2baa249395d3ef/Woodworking/Opc.Ua.Eumabois.Nodeset2.xml#L14

Downloading the file from https://files.opcfoundation.org/schemas/Woodworking/1.0/Opc.Ua.Eumabois.Nodeset2.xml shows the same thing.

https://github.com/OPCFoundation/UA-Nodeset/commit/6ab730079bd80c3ef572c3f9f80b76e94df5449e shows 1.00 release date "2021-10-03"
[OPC UA Reference](https://reference.opcfoundation.org/Woodworking/v100/docs/) shows 1.00 release date "2021-10-12"

Is the subminor/last number in the version supposed to be ignored?

jacobq, 2023-04-06:
Per the GH issue template suggestion, I created an issue for this in the Mantis issue tracker: [#8785](https://mantis.opcfoundation.org/view.php?id=8785)

GoetzGoerisch, 2023-04-07:
Hi @jacobq

the Woodworking Specification release date is 2021-10-03:

[image]

@randy-armstrong why is the online reference listing a different date?

The [Woodworking Example](https://github.com/OPCFoundation/UA-Nodeset/blob/latest/Woodworking/Opc.Ua.Eumabois.Nodeset2.xml) has unfortunately a leftover date from the prototyping phase. Thus the mismatch in the required models section.

You just can replace the version 1.00.8 with 1.00.0 for trying this sample nodeset.

Thank you for creating the mantis issue. @ThomasRothfuss please have a look with the editor team.
A fix is already in progress for the release of v1.01.0 of woodworking, as far as I know.

Another demo nodeset is here: https://github.com/umati/Sample-Server-node-opcua/blob/develop/models/basyskom_woodworking_instance.xml

Please let me know if you need more information.

opcfoundation-org, 2023-04-07:
OnlineReference gets the date from here:
https://profiles.opcfoundation.org/document/69

jacobq, 2023-04-07:
OK, I will close this now since you have confirmed that this is a mistake that can be safely ignored. I recently started diving into OPC UA for the first time so don't have enough experience to recognize what is an error in existing artifacts vs. what is an error in my own understanding.

closed 2023-04-07 by jacobq