OPCFoundation/UA-Nodeset #95: BOM problem

undroidwish, 2023-07-31:
[UA-Nodeset](https://github.com/OPCFoundation/UA-Nodeset/tree/latest)/[Robotics](https://github.com/OPCFoundation/UA-Nodeset/tree/latest/Robotics)

The file OpcUaRoboticsModel.xml is not a valid XML document since it contains a UTF-8 encoded BOM at its start.

opcfoundation-org, 2023-07-31:
Do you have a reference to a spec that says a BOM is not valid XML or is it an issue with the tool you want to use?

opcfoundation-org, 2023-07-31:
Also OpcUaRoboticsModel.xml is not a normative document and will be deleted in the future. 
You should be able to strip the BOM with any tool like Notepad++.

GoetzGoerisch, 2023-08-01:
`OpcUaRoboticsModel.xml` is the input file for the ModelCompiler, added by the Editor to work with the .Net SDK.

@xenonforlife

opcfoundation-org, 2023-08-01:
The ModelCompiler can accept the normative NodeSet file as input now. A separate ModelDesign is not required.
That said, the ModelCompiler should be able to deal with the BOM.

Checked in a version with no BOM.

closed 2026-10-21 by opcfoundation-org