OPCFoundation/UA-Nodeset #98: Question: Defining RolePermissions, UserRolePermissions and AccessRestrictions in nodeset file

DStr276, 2023-09-21:
We've used the UA Modeler by Unified Automation to add the specification for RolePermissions, UserRolePermissions and AccessRestrictions to the nodeset file. It seems that this extension might only work in combination with the SDKs provided by Unified Automation and not with other OPC UA server projects such as node opcua.
We've also tried to add these elements to some nodes innaccordance to the xsd file - no success.

How does one define these elements correctly in the nodeset file?  
BR

opcfoundation-org, 2023-09-21:
UserRolePermissions are dynamic and do not belong in a persistent file.

These attributes were added to 1.04 so some SDKs may not have added support.
You need to contact the maintainers of node opcua and ask when they will add support for the 1.04 attributes.

DStr276, 2023-09-21:
Thank you for the quick response!

closed 2026-10-24 by opcfoundation-org