OPCFoundation/UA-Nodeset #99: Robotics/Opc.Ua.Robotics.NodeSet2.xml - namespace version & PublicationDate not updated in the Server / Namespaces section

plamenk, 2023-09-27:
This causes problems with loading into some OPC UA server. Concrete example: the Python one, https://github.com/FreeOpcUa/opcua-asyncio. Attached the modified file with the two corrections.
Plamen
[Opc.Ua.Robotics.NodeSet2_mod.zip](https://github.com/OPCFoundation/UA-Nodeset/files/12743823/Opc.Ua.Robotics.NodeSet2_mod.zip)

BFrKUKA, 2023-10-09:
Can you please describe the two corrections by naming the line number or copying the changed corrections?
Because until now there have not been problems to use the nodeset with SDKs used by several companies.
Thanks

plamenk, 2023-10-09:
Here the changes from my repo commit:
@@ -111,7 +111,7 @@
      <Reference ReferenceType="HasProperty" IsForward="false">ns=1;i=15011</Reference>
    </References>
    <Value>
      <String xmlns="http://opcfoundation.org/UA/2008/02/Types.xsd">1.01</String>
      <String xmlns="http://opcfoundation.org/UA/2008/02/Types.xsd">1.01.2</String>
    </Value>
  </UAVariable>
  <UAVariable NodeId="ns=1;i=15091" BrowseName="NamespacePublicationDate" ParentNodeId="ns=1;i=15011" DataType="DateTime">
@@ -121,7 +121,7 @@
      <Reference ReferenceType="HasProperty" IsForward="false">ns=1;i=15011</Reference>
    </References>
    <Value>
      <DateTime xmlns="http://opcfoundation.org/UA/2008/02/Types.xsd">2020-05-20T00:00:00Z</DateTime>
      <DateTime xmlns="http://opcfoundation.org/UA/2008/02/Types.xsd">2021-05-20T00:00:00Z</DateTime>
    </Value>
  </UAVariable>
  <UAVariable NodeId="ns=1;i=15114" BrowseName="IsNamespaceSubset" ParentNodeId="ns=1;i=15011" DataType="Boolean">

BFrKUKA, 2023-10-09:
sorry - now i am confused - here the linkt to the latest robotics nodeset:
https://github.com/OPCFoundation/UA-Nodeset/blob/latest/Robotics/Opc.Ua.Robotics.NodeSet2.xml

The version and PublicationDate is as you mentioned in your modified nodeset:
<Model ModelUri="http://opcfoundation.org/UA/Robotics/" Version="1.01.2" PublicationDate="2021-05-20T00:00:00Z">

What is the topic about ns=1;i=15011? 

What is meant by @@ xxx - xxx @@?

plamenk, 2023-10-09:
I am sorry for the confusion. The above extract was the comparison of the two versions after my correction. The two problematic places are the Version and PublicationDate from the Server -> Namespaces section, marked below from the current Robotics nodeset:
...
  <UAVariable NodeId="ns=1;i=15064" BrowseName="NamespaceVersion" ParentNodeId="ns=1;i=15011" DataType="String">
    <DisplayName>NamespaceVersion</DisplayName>
    <References>
      <Reference ReferenceType="HasTypeDefinition">i=68</Reference>
      <Reference ReferenceType="HasProperty" IsForward="false">ns=1;i=15011</Reference>
    </References>
    <Value>
      <String xmlns="http://opcfoundation.org/UA/2008/02/Types.xsd">**1.01**</String>
    </Value>
  </UAVariable>
  <UAVariable NodeId="ns=1;i=15091" BrowseName="NamespacePublicationDate" ParentNodeId="ns=1;i=15011" DataType="DateTime">
    <DisplayName>NamespacePublicationDate</DisplayName>
    <References>
      <Reference ReferenceType="HasTypeDefinition">i=68</Reference>
      <Reference ReferenceType="HasProperty" IsForward="false">ns=1;i=15011</Reference>
    </References>
    <Value>
      <DateTime xmlns="http://opcfoundation.org/UA/2008/02/Types.xsd">**2020-05-20T00:00:00Z**</DateTime>
    </Value>
  </UAVariable>
...

They do not match the correct ones, 1.0.1.2 and 2021-05-20T00:00:00Z respectively, of the nodeset. After debugging the OPC UA Python server opcua-asyncio, I realized that the actual version and dates are taken from there. 

I hope that this is clear now.

Plamen

BFrKUKA, 2023-10-10:
so this issue can be closed, because the Python software does not work well and the Robotics nodeset has no issue?

plamenk, 2023-10-10:
as you formulate it as a question, here my answer: it could be that the opcua-asyncio server has an unusual interpretation of the namespace dependencies, nevertheless, only the Robotics nodeset shows this difference between NamespacePublicationDate / NamespaceVersion respectively PublicationDate / Version tags. I would fix it.

BFrKUKA, 2023-10-10:
So in the latest Robotics nodeset it is ok. Because the version and date are correct:
https://github.com/OPCFoundation/UA-Nodeset/blob/latest/Robotics/Opc.Ua.Robotics.NodeSet2.xml
 Model ModelUri="http://opcfoundation.org/UA/Robotics/" Version="1.01.2" PublicationDate="2021-05-20T00:00:00Z"

You created a new issue for the Python Server : https://github.com/FreeOpcUa/opcua-asyncio/issues/1464.
I think it is the needed to fix it there - so this issue (to change the Robotics nodeset) can be closed?

plamenk, 2023-10-10:
I don’t think so. Where is the problem to bring the Namespace* tags from lines 114 and 124 in sync with line 37 in the modeste file?

BFrKUKA, 2023-10-10:
Thanks - now I have seen it - ich forked and changed the lines... and started a pull request to get it final

closed 2026-10-21 by opcfoundation-org