OPCFoundation/UA-Nodeset #100: opc.ua.i4aas.nodeset2.xml - Line: 1818: "AccessPermissionRule_0" error

engacosta, 2023-09-30:
Hello, how are you?

I am trying to import opc.ua.i4aas.nodeset2.xml but following error is indicated:

XML Parsing - File: opc.ua.i4aas.nodeset2.xml - Line: 1818 - Parsing resulted in error : "BadInvalidArgument" - for Element/Attibute Name: "Type" - Value: "AccessPermissionRule_0"Please correct errors and upload file again.
The following syntax / schema errors occured : 
Failed to Import File object details in Server. Error: BadSyntaxError: A value had an invalid syntax.(0x80B60000)

Do you have a solution suggestion since I did not find nothing about in the documentation?

Thank you!

opcfoundation-org, 2023-09-30:
What tool are you using to import?
Seems like the tool does not support the latest NodeSet schema.

engacosta, 2023-09-30:
> What tool are you using to import? Seems like the tool does not support the latest NodeSet schema.
Hi OPCF! Thank you for the quick reply.

I am using Matrikon Data Broker and I was able to import most of the the v1.05 and 1.04 Nodesets.

I also tried to use Unified Automation UAModeler with the same XML Nodeset and the software screen freeze on format conversion step (using cloned file from Github Desktop). It works as expected if I save a raw XML file, however I understand that  UAModeler does not check the XML inconsistencies.

opcfoundation-org, 2023-09-30:
The I4AAS nodeset does not have the embedded XSD needed to parse the values using DataTypes defined in the NodeSet.
This is allowed in 1.04 but it is possible existing tools are failing because they are looking for the embedded XSD.

You should contact Matrikon and ask them if their tools support NodeSets that only support the DataTypeDefinition Attribute which was introduced in 1.04.

engacosta, 2023-09-30:
> The I4AAS nodeset does not have the embedded XSD needed to parse the values using DataTypes defined in the NodeSet. This is allowed in 1.04 but it is possible existing tools are failing because they [...]

Thank you very much for your support! I will contact Matrikon support. 
"Ótimo final de semana!" (Have a great weekend in Brazilian portuguese)!

BFrKUKA, 2023-10-09:
Hello engacosta,
please be aware, that the I4AAS Companion Specification is outdated, because it is NOT based on the latest "AAS Metamodel V3" (AAS Part 1)  which was released in spring this year. (OPCUA I4AAS CS is several years old and not compatible to ne AAS Metamodel V3).
There are discussions if there will be an "I4AAS V3 CS" development. 
Today AAS V3 is based on the updated metamodel (AAS Part 1)  and the RestApi Interface (Part 2)- see here for latest specifications of AAS:
https://industrialdigitaltwin.org/en/content-hub/downloads

closed 2026-10-21 by opcfoundation-org