OPCFoundation/UA-Nodeset #108: BrowseName missing namespace index

JElgar, 2025-01-10:
Ive notice in some specs the BrowseName is missing a namespace index e.g.

https://github.com/OPCFoundation/UA-Nodeset/blob/dee199135d85a5cb0dc5db25e27228b1f9d7ea0d/AutoID/Opc.Ua.AutoID.NodeSet2.xml#L173C36-L173C59

According to this: https://reference.opcfoundation.org/Core/Part6/v104/docs/F.3 it seems like both the namespace index and the name should be included.

This may be related to this issue but its not totally clear to me https://github.com/OPCFoundation/UA-Nodeset/issues/22

Im trying to understand

1. Why is there this discrepancy here, am I missing something?
2. Is there some fallback value for when the namespace index is not there (e.g. is it always 0 as this seem to suggest https://reference.opcfoundation.org/DI/v102/docs/11.2)?

AndreasHeine, 2025-01-10:
https://reference.opcfoundation.org/Core/Part3/v104/docs/5.2.4

Servers may often choose to use the same namespace for the NodeId and the BrowseName. However, if they want to provide a standard Property, its BrowseName shall have the namespace of the standards body although the namespace of the NodeId reflects something else, for example the local Server.

JElgar, 2025-01-10:
Thank you for the quick response! Sorry I find that paragraph quite hard to understand, does that mean if a BrowseName does not have a namespace index then the namespace index matches that of the NodeId?

AndreasHeine, 2025-01-10:
> Thank you for the quick response! Sorry I find that paragraph quite hard to understand, does that mean if a BrowseName does not have a namespace index then the namespace index matches that of the NodeId?

its a companion spec. so it shall have the index of the origin standard body so either di or opc ua... 

still looking!

JElgar, 2025-01-10:
Sorry I'm not very familiar with this stuff so still a bit lost. Why does that Alias mean it comes from ua namespace?

AndreasHeine, 2025-01-10:
https://reference.opcfoundation.org/Core/Part6/v104/docs/F.15

maybe this section helps

opcfoundation-org, 2025-01-10:
The string syntax is defined here:
https://reference.opcfoundation.org/Core/Part6/v105/docs/5.1.12#_Ref122649467

The NamespaceIndex is not missing. It is omitted if NS=0
i.e. 0:EnumValues and EnumValues are the same.

JElgar, 2025-01-20:
Ah thats brill thank you!

Does this mean the documentation in other places needs updating, or are those referring to a different kind of qualified name? e.g. both the following don't mention its optional
- https://reference.opcfoundation.org/Core/Part6/v105/docs/F.3
- https://reference.opcfoundation.org/Core/Part3/v105/docs/8.3

opcfoundation-org, 2025-01-21:
The first reference should reference the formal syntax that was not defined when the text was written.

The second reference is the encoding independent definition of the QualifiedName which always has the NamespaceIndex.

closed 2026-10-21 by opcfoundation-org