OPCFoundation/UA-Nodeset #110: load nodeset2.xml to python opcua server

hadifar, 2025-01-20:
How can I import nodeset2.xml (e.g., https://github.com/OPCFoundation/UA-Nodeset/blob/latest/PackML/Opc.Ua.PackML.NodeSet2.xml) using python-OPC-UA? 

Here is my code:

```
from opcua import Server

file_dir = "Opc.Ua.PackML.NodeSet2.xml"
server = Server()
server.import_xml(file_dir)
server.start()
```

It gives the following error:

```
Traceback (most recent call last):
  File "C:\Users\IdeaProjects\wind-turbine-poc\experimental\load_nodeset2.py", line 16, in <module>
    main()
  File "C:\Users\\IdeaProjects\wind-turbine-poc\experimental\load_nodeset2.py", line 11, in main
    server.import_xml('C:/Users//IdeaProjects/FAPSDemoOPCUAServer2/UA-Nodeset-master/PackML/Opc.Ua.PackML.NodeSet2.xml')
  File "C:\Users\\IdeaProjects\wind-turbine-poc\venv\lib\site-packages\opcua\server\server.py", line 518, in import_xml
    return importer.import_xml(path, xmlstring)
  File "C:\Users\\IdeaProjects\wind-turbine-poc\venv\lib\site-packages\opcua\common\xmlimporter.py", line 68, in import_xml
    node = self._add_node_data(nodedata)
  File "C:\Users\\IdeaProjects\wind-turbine-poc\venv\lib\site-packages\opcua\common\xmlimporter.py", line 89, in _add_node_data
    node = self.add_variable_type(nodedata)
  File "C:\Users\\IdeaProjects\wind-turbine-poc\venv\lib\site-packages\opcua\common\xmlimporter.py", line 330, in add_variable_type
    if obj.value and len(obj.value) == 1:
TypeError: object of type 'ExtObj' has no len()
```

AndreasHeine, 2025-01-20:
[image]

the old python-opcua library is not maintained anymore and so forth it does not support Structures that well!

m-obi, 2025-01-21:
Did you look the examples server-woodworking.py and server-robotics.py?

hadifar, 2025-01-23:
@m-obi Thanks it works.

https://github.com/FreeOpcUa/opcua-asyncio/blob/master/examples/server-woodworking.py

closed 2025-01-23 by hadifar