OPCFoundation/UA-Nodeset #112: Getting NodeState for i=68 (PropertyType) returns an object of type BaseDataVariableTypeState

wbtoms, 2025-01-27:
When getting the NodeState corresponding to the node PropertyType (i=68) using the method FindNodeInAddressSpace of class CustomNodeManager2, the NodeState object returned is BaseDataVariableTypeState instead of a PropertyTypeState.

This seems to be related to the loading of predefined node from the binary file Opc.Ua.Stack.Generated.Opc.Ua.PredefinedNodes.uanodes in the DiagnosticsNodeManager, see attached screenshot.

To reproduce: Use a reference server and just call FindNodeInAddressSpace on the node manager using the node ID i=68.

Can be reproduced with UA-.NETStandard release version 1.5.374.158 or with master branch (at least until end of December 2024).

[image]

opcfoundation-org, 2025-01-27:
This is an artifact of the untyped import mechanism (i.e. all nodes are imported as instances of the base class and it is up to the subclass to replace them with specific subtypes in the CreateAddressSpace() method override).

All VariableTypes are imported as BaseDataVariableTypes which is correct for every VariableType except PropertyType.

It could be fixed in the DiagnosticsNodeManager.CreateAddressSpace() method (need a issue against the NETStandard GitHub).

However, the question becomes: why does it matter?

wbtoms, 2025-01-28:
Yes I did not put this bug on the right project, I will open it in the UA-.NetStandard project.

Well, it does matter because:
1. it is the wrong type obvioulsy, therefore you cannot rely on the loaded nodes.
2. the test `node is PropertyTypeState' does not work.

closed 2025-01-28 by wbtoms