OPCFoundation/UA-Nodeset #115: Best practice for extending companion specification

kristianmo, 2025-07-03:
Hello, there is a companion specification (CS) with types defined in namespace 2. Instances of each CS type are added in a namespace 3. The variables in the types are all missing the Definition property. Until the companion specification gets an update, what is the best practice for extending/adding Definition or other properties to variables and objects in an CS. 

- Should the types get new optional properties in a new namespace?

- Should the types be left alone and add the new properties to the instances in namespace 3?

- Should the types be left alone and add the new properties to the instances in a new namespace 4?

- Other options?

opcfoundation-org, 2025-07-03:
Definition property is only used for DataTypes. Can you clarify?

kristianmo, 2025-07-03:
The CS used for instance TwoStateDicreteType for bool in the types. They did not include Definition, or more generally if I need to add MySuperImportantProperty.

[image]

closed 2026-10-24 by opcfoundation-org