OPCFoundation/UA-Nodeset #116: Typo in opcua-statuscodes.ts

pc2005cz, 2025-09-16:
Hellom
it seems there is a typo "Sempahore" (instead Semaphore) at this line https://github.com/OPCFoundation/UA-Nodeset/blob/c8368defcd51d96a74c557f992cb7b54b93a1773/TypeScript/opcua-statuscodes.ts#L202C4-L202C27

Other instances of this typo were fixed in this patch https://github.com/OPCFoundation/UA-Nodeset/commit/f11293ad04379bd6918dbe55fea669933948fcef

opcfoundation-org, 2025-09-16:
Fixed in latest version:
https://github.com/OPCFoundation/UA-Nodeset/blob/c8368defcd51d96a74c557f992cb7b54b93a1773/OpenApi/Constants/TypeScript/opcua_statuscodes.ts#L105

pc2005cz, 2025-09-16:
Ah I see there are two versions of opcua-statuscodes.ts vs opcua_statuscodes.ts in the repo, sorry. 

(I was trying to check if open62541 library https://github.com/open62541/open62541/pull/7274 is already updated so I've run a simple git grep on this repo)

closed 2025-09-16 by pc2005cz